mortenfyhn/coffee-scales #30: Design new PCB

mortenfyhn, 2020-11-01:


sindrehan, 2020-11-20:
Let me know if you want a review of/help with the PCB. Are you using KiCad?

mortenfyhn, 2020-11-20:
That'd be great! @yttersian is helping out with PCB design, maybe we can all collaborate? I don't remember what he suggested using. @Slattsveen has also offered to help with making an enclosure / machining.

Roughly, the plan for the first pcb revision is

* Still use the full arduino nano 328, and just solder its pin headers to the pcb.
* Get rid of the seven segment display breakouts, get the TM1637 + display circuits onto our board
* Add batteries + battery circuit (rechargeable, or disposable if they last long enough)
* Add a tare button
* Either create a drop-in slot for the HX711 breakout, or re-implement that directly on the board

closed 2023-04-30 by mortenfyhn